WorldFamousElectronics/PulseSensorPlayground #146: Adafruit Feather M0 WiFi with uFL - ATSAMD21 + ATWINC1500 - fw 19.4.4

awatterson22, 2022-02-08:
Hi, I am using this board, and I am having trouble getting the PulseSensor_BPM to work

awatterson22, 2022-02-08:
This is what I get when I try to run PulseSensor_BPM_Alternative 

In file included from /Users/abbywatterson/Documents/Arduino/libraries/PulseSensor_Playground/src/PulseSensorPlayground.h:439,
                 from /Users/abbywatterson/Documents/Arduino/libraries/PulseSensor_Playground/examples/PulseSensor_BPM_Alternative/PulseSensor_BPM_Alternative.ino:41:
/Users/abbywatterson/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundDisableInterrupt()':
/Users/abbywatterson/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/Interrupts.h:276:1: error: no return statement in function returning non-void [-Werror=return-type]
  276 | }
      | ^
In file included from /Users/abbywatterson/Documents/Arduino/libraries/PulseSensor_Playground/src/PulseSensorPlayground.h:439,
                 from /Users/abbywatterson/Documents/Arduino/libraries/PulseSensor_Playground/examples/PulseSensor_BPM_Alternative/PulseSensor_BPM_Alternative.ino:41:
/Users/abbywatterson/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundEnableInterrupt()':
/Users/abbywatterson/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/Interrupts.h:321:1: error: no return statement in function returning non-void [-Werror=return-type]
  321 | }
      | ^
cc1plus: some warnings being treated as errors
exit status 1
Error compiling for board Adafruit Feather M0 (SAMD21).

biomurph, 2022-02-08:
@awatterson22 
It looks from your the error that you post that the library has a bug in it.
I will dig into this and fix it. Should be short fix.

biomurph, 2022-02-08:
@awatterson22 
Please update your PulseSensor Playground library in Arduino.
The latest version is v1.5.2
It might not show up in the Arduino IDE drop down libraries tool. If you don't see it, please download the ZIP of this repo and install it into your Arduino/libraries folder.

Let us know if you have any further issues, or if this fixes your problem.
I was unable to replicate your issue on my end.

awatterson22, 2022-02-08:
Hi! So I tried to download the zip and re-upload the library but I am still getting the same error. I looked through the utility functions and it doesn't look like there is a SAMD board, just AVR.

biomurph, 2022-02-08:
Oh, it may have to do with the example you're using.
The Feather M0 with the WiFi does not currently support hardware interrupts.
Please use the example PulseSensor_BPM_Alternative.ino instead.

awatterson22, 2022-02-08:
Ah okay, will do - I just have one other question. For the PulseSensor_BPM_Alternative.ino, is the first number BPM and the second IBI? What is the third number?

biomurph, 2022-02-08:
The library prints out the values of the BPM, the IBI, and also the raw sensor value. That's the third number. 
You can visualize them all with the Arduino Serial Plotter.

awatterson22, 2022-02-17:
Hi! Thank you! I got everything set up and it works well. Is there a way to get the sample output to return to a variable rather than printing on the serial monitor?

biomurph, 2023-10-26:
PulseSenor Playground Library supports SAMD based boards

closed 2023-10-26 by biomurph