WorldFamousElectronics/PulseSensorPlayground #147: Generating RMSSD and SDNN from IBI, BPM data

sumamovva, 2022-03-09:
Hi, I am trying to calculate the RMSDD and SDNN statistics after collecting IBI, BPM data using your library with an Arduino Uno. I was digging around online and actually came across an old commit of your (now) _deprecated_ library, where someone seems to be doing these calculations: https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/pull/15/files

Is there a reason why these changes were not approved? For your reference, I tried modifying the library myself with similar changes, but it seems that RMSDD and SDNN both output 0.0, so the code modifications I made likely have a bug: https://github.com/awatterson22/hrv-analysis/tree/master/PulseSensor_Playground

Do you know how to approach this issue or if you have ideas on how to more simply calculate RMSDD and SDNN using this library?

Thanks!

biomurph, 2022-03-09:
@sumamovva 
I can't give you a good reason why the mods were not approved. We are not making an HRV specific device.
We have some HRV specific code written for Processing [here](https://github.com/WorldFamousElectronics/PulseSensor_Heart-Rate-Variabilty).

That said, I would think that since you are relying on the IBI values to determine the HRV, and thereby derive the relative change in beat timing, that you could do your calculations in the .ino file. No need to hack into the library, really, in order to get the basic result that you want. 

Can you share the algorithm and maths that you're using?

sumamovva, 2022-03-22:
Hi Joel,

Sorry for the late reply - we were off on break! This is the general
algorithm that we plan to use (attached below).

Thank you so much for your help!

Best,
Suma

closed 2023-10-26 by biomurph