WorldFamousElectronics/PulseSensorPlayground #148: Adafruit Feather nRF52832 Compatibility?

surffl, 2022-03-10:
Hello,

I have been trying to get the PulseSensor_BPM_Alternative example to work with my Adafruit Feather nRF52832 board but have been unsuccessful. I have read through the other issue here and installed the latest version of the library from here. I continue to get the following error:

`In file included from /Users/mateo/Documents/Arduino/libraries/PulseSensorPlayground-master/src/PulseSensorPlayground.h:439,
                 from /Users/mateo/Documents/Arduino/libraries/PulseSensorPlayground-master/examples/PulseSensor_BPM_Alternative/PulseSensor_BPM_Alternative.ino:41:
/Users/mateo/Documents/Arduino/libraries/PulseSensorPlayground-master/src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundDisableInterrupt()':
/Users/mateo/Documents/Arduino/libraries/PulseSensorPlayground-master/src/utility/Interrupts.h:297:1: error: no return statement in function returning non-void [-Werror=return-type]
  297 | }
      | ^
In file included from /Users/mateo/Documents/Arduino/libraries/PulseSensorPlayground-master/src/PulseSensorPlayground.h:439,
                 from /Users/mateo/Documents/Arduino/libraries/PulseSensorPlayground-master/examples/PulseSensor_BPM_Alternative/PulseSensor_BPM_Alternative.ino:41:
/Users/mateo/Documents/Arduino/libraries/PulseSensorPlayground-master/src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundEnableInterrupt()':
/Users/mateo/Documents/Arduino/libraries/PulseSensorPlayground-master/src/utility/Interrupts.h:349:1: error: no return statement in function returning non-void [-Werror=return-type]
  349 | }
      | ^
cc1plus: some warnings being treated as errors
exit status 1
Error compiling for board Adafruit Feather nRF52832.`

Any chance you have any advice/a fix? Thanks!

biomurph, 2022-03-10:
@surffl 
I compiled the BPM Alternative example on two machines here with no errors.
Then, I found that the Arduino Libraries Manager was not offering the latest version.
I dug into the Arduino Libraries Manager requirements, and discovered a space ' ' that was preventing ALM from updating.

That bug is now fixed, and the Libraries manager tool will scrape later today. We are up to v1.6.0
You can also go to the releases (lower right side bar) and download a zip of the latest version.

surffl, 2022-03-10:
@biomurph 

Just deleted the old library, downloaded the zip of your newest release here, and re-added the library, but unfortunately I am getting exactly the same error when trying to compile the BPM alternative :/

biomurph, 2022-03-10:
@surffl I can't replicate your problem.
What versions of all the softwares are you using?
What is your OS?
what version Adafruit board files are you using?

biomurph, 2023-10-26:
PulseSenor Playground Library supports nRF52 boards