WorldFamousElectronics/PulseSensorPlayground #153: USE_ARDUINO_INTERRUPTS problem with NodeMCU 0.9

Abdelrahman-Kh-Fouad, 2022-06-29:
I tried to just include PulseSensorPlayground with `USE_ARDUINO_INTERRUPTS false`, but every time complier gave me this error!
```text
in file included from .pio/libdeps/nodemcu/PulseSensor Playground/src/PulseSensorPlayground.h:439,
                 from src/main.cpp:41:
.pio/libdeps/nodemcu/PulseSensor Playground/src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundDisableInterrupt()':
.pio/libdeps/nodemcu/PulseSensor Playground/src/utility/Interrupts.h:297:1: error: no return statement in function returning non-void [-Werror=return-type]
  297 | }
      | ^
In file included from .pio/libdeps/nodemcu/PulseSensor Playground/src/PulseSensorPlayground.h:439,
                 from src/main.cpp:41:
.pio/libdeps/nodemcu/PulseSensor Playground/src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundEnableInterrupt()':
.pio/libdeps/nodemcu/PulseSensor Playground/src/utility/Interrupts.h:349:1: error: no return statement in function returning non-void [-Werror=return-type]
  349 | }
      | ^
cc1plus: some warnings being treated as errors
*** [.pio/build/nodemcu/src/main.cpp.o] Error 1
```
but with `USE_ARDUINO_INTERRUPTS true` i could include the library and when pulseSensor.begin() invoked it didn't return true.

biomurph, 2022-06-29:
@Abdelrahman-Kh-Fouad 
Our library has two options to use hardware timer interrupts or to use software interrupts.
Which example code are you using?
I am going to try and replicate your problem in my lab and get back to you on this thread.

Abdelrahman-Kh-Fouad, 2022-06-29:
@biomurph i tried to use software interrupts, but the problem on including i think.

biomurph, 2022-06-29:
@Abdelrahman-Kh-Fouad 
I don't have a Node MCU 0.9 board here in my lab, but I was able to test by compiling.
I have the latest `ES8266 Boards v3.0.2` in my Boards Manager, and it lists the `Node MCU 0.9` as an option.
When I select it and compile, I get the following results

```
Example PulseSensor_BPM.ino
#define USE_ARDUINO_INTERRUPTS true
```
With this sketch and #define setting, I get the 'return false' error.

```
Example PulseSensor_BPM_Alternative.ino
#define USE_ARDUINO_INTERRUPTS false
```

With this example, I get a clean compile. Please try using PulseSensor_BPM_Alternative.ino sketch and don't change anything in the example.

We use a hardware timer to get the most accurate sample rate. For some platforms, we don't support Timer Interrupts, so for those you need to use our "Alternative" Sketch examples to get started.

closed 2022-06-29 by biomurph