WorldFamousElectronics/PulseSensorPlayground #154: GETTING STARTED WORKING BUT NOT GETTING BPM TO MONITOR

finntuberlin, 2022-07-12:
Hello All,

When I run the getting started example sketch I can see a pulse signal in my serial plotter, which tells me my sensor is working, however, when I open the getting_bpm_to_monitor sketch, all I see is random inconsistent values that only show up if I lift my finger up and down quickly, otherwise if a keep my finger on the sensor no values are shown. I am using the goldilocks amount of pressure. I have also played around with the threshold value to no avail. The code hasn't been changed at all from the example, other than playing with the threshold value, but I have kept the threshold value within the recommended window. 

Thank you

biomurph, 2022-07-12:
@finntuberlin 
The problem you are having is straightforward.
The Serial Plotter in Arduino is expecting to receive some number, or a series of numbers, separated by a comma or a space or a tab, and terminating in a carriage return.
The Getting_BPM_to_Monitor.ino sketch is not doing that. If you look into the loop function, that sketch is sending human readable text that is not formatted to the Serial Plotter. The Serial Plotter is doing it's best to try and interpret the data, and that's why you're seeing all the crazy nonsense.

Instead, open up the Serial Monitor, and you will see what the sketch is sending.

finntuberlin, 2022-07-12:
Yes I mean Arduino UNO sorry about the mixup. 

Not sure with Arduino_BPM if you meant a screen shot of the Getting_BPM_to_Monitor sketch (1st photo) or the Pulsesensor_BPM sketch (2nd photo) but here are screenshots from the Serial Plotter from both. Also attached is a photo of my setup.

[image]
[image]
[image]

biomurph, 2022-07-12:
@finntuberlin 
I see the problem.
The sensor you are using is not our Pulse Sensor. It's clear from the picture of the hardware.
Also note in [this guide](https://pulsesensor.com/pages/getting-advanced) at the bottom there is a screen shot of the Serial Plotter. You should be seeing a very big pulse signal. The one you show is very small.

Please support the original makers of Pulse Sensor by purchasing the legit product.
Our Pulse Sensor is available on our website, [pulsesensor.com](https://pulsesensor.com/), as well as
Adafruit
Sparkfun
Digikey
Mouser

Also, in some regions of the world, you can find Pulse Sensor available in brick and mortar stores, as Adafruit and Sparkfun bundle products for distribution to resellers.

closed 2023-10-26 by biomurph