WorldFamousElectronics/PulseSensorPlayground #157: Getting_BPM_to_monitor sketch displaying random values at 11250 baud

NotAvalibleLol, 2022-10-22:
The Getting_BPM_to_monitor sketch displays random values in commas, an example would be 0,360,970.
[image]
Any other baud level, higher or lower, displays random characters, both on the Arduino 2.0.0 IDE and the Arduino 1.8.19 legacy version.
[image]
[image]
I am using an Arduino UNO R3, and it is on the COM 3 port.

biomurph, 2022-10-25:
@NotAvalibleLol 
I cannot replicate your problem on the Uno R3.
Please make sure that you upload the sketch to the Arduino, and that your baud rate is set to 9600.

NotAvalibleLol, 2022-10-26:
Thanks for getting back to me @biomurph,

I switched to 9600 baud, but now it shows random characters.
[image]

NotAvalibleLol, 2022-10-30:
No, any code that I try to put in will always have the same output, random characters.

NotAvalibleLol, 2022-10-30:
I managed to fix it, since my sketches were not uploading properly, but now it shows very high values, upwards of 200.
[image]

NotAvalibleLol, 2022-11-04:
Thanks for replying!

Here is a photo of my setup, it is fairly simple. There is no LED or breadboard since I am just testing out the sketches. The Arduino UNO
[image]
 is off in this photo.

closed 2023-10-26 by biomurph