WorldFamousElectronics/PulseSensorPlayground #158: Node MCU V1 - Plotter is working but no BPM in Monitor

gleameyes10, 2022-11-09:
I am using a node mcu v1 esp8266. ECG on serial plotter is very clear, but it does not show BPM in serial monitor. I tried the GettingBPMtoMonitor and PulseSensor BPM alternative but it does now show BPM. I need help for my capstone.

biomurph, 2022-11-09:
@gleameyes10 
Well, it sounds like you chose a nice challenge for your capstone!

The bare ESP8266 pin can only take 1V maximum, but it looks like the Node MCU version has an on-board voltage divider for you.

It will help to troubleshoot if you can answer these questions.
What is the OS you are using?
What sketch are you using to try and get BPM?
Also, upload a picture of your hardware so that I can verify that it is legit.

gleameyes10, 2022-11-13:
I was using GettingBPMtoMonitor sketch under the pulse sensor play ground example.  I used DFROBOT Ecg sensor because the project needs to work while moving. I've tried to use this sketch and it seems to be working well. 

void setup() {
  // put your setup code here, to run once:
Serial.begin (9600);
}

void loop() {
  // put your main code here, to run repeatedly:
Serial.println(analogRead(A0));
}

I have an arduino uno and this sketch (GettingBPMtoMonitor) seems to be working.
I'm using windows 10

biomurph, 2022-11-14:
@gleameyes10 
first thing, it is not an ECG sensor. it is a PPG sensor.

I cannot provide support or troubleshooting if you are using cloned hardware.

closed 2023-10-26 by biomurph