WorldFamousElectronics/PulseSensorPlayground #161: Code error for sp32 pulse sensor data to thinspeak

abdulmoizs502, 2023-01-21:
I'm making an esp32 pulse sensor and facing problems with the code to send my pulse sensor data to thingspeak. Here's the code I'm using:

```

#define  USE_ARDUIO_INTERRUPTS true

#include "ThingSpeak.h"

#include <WiFi.h>

#include <PulseSensorPlayground.h>

const char* ssid = "HUAWEI-ChwU";   // your network SSID (name)

const char* password = "03003038745";   // your network password

WiFiClient  client;

unsigned long myChannelNumber = 2010078 ;

const char * myWriteAPIKey = "RHRTPZAXDHUT927V";

// Timer variables

unsigned long lastTime = 0;

unsigned long timerDelay = 30000;

//  Variables

const int PulseWire = 36;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0

const int LED = 2;          // The on-board Arduino LED, close to PIN 13.

int Threshold = 2000;           // Determine which Signal to "count as a beat" and which to ignore.

                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.

                               // Otherwise leave the default "550" value.

PulseSensorPlayground pulseSensor;   // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void initBPM() {  

  // Configure the PulseSensor object, by assigning our variables to it.

  pulseSensor.analogInput(PulseWire);  

  pulseSensor.blinkOnPulse(LED);       //auto-magically blink Arduino's LED with heartbeat.

  pulseSensor.setThreshold(Threshold);  

  // Double-check the "pulseSensor" object was created and "began" seeing a signal.

   if (pulseSensor.begin()) {

    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  

  }

}

void setup() {

  Serial.begin(9600);          // For Serial Monitor

  initBPM();

  WiFi.mode(WIFI_STA);  

  ThingSpeak.begin(client);  // Initialize ThingSpeak

}

void loop() {

  if ((millis() - lastTime) > timerDelay) {

    // Connect or reconnect to WiFi

    if(WiFi.status() != WL_CONNECTED){

      Serial.print("Attempting to connect");

      while(WiFi.status() != WL_CONNECTED){

        WiFi.begin(ssid, password);

        delay(5000);    

      }

      Serial.println("\nConnected.");

    }

    int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".

                                               // "myBPM" hold this BPM value now.

   if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened".

   Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".

   Serial.print("BPM: ");                        // Print phrase "BPM: "

   Serial.println(myBPM);                        // Print the value inside of myBPM.

   }

   // Write to ThingSpeak. There are up to 8 fields in a channel, allowing you to store up to 8 different

    // pieces of information in a channel.  Here, we write to field 1.

    int x = ThingSpeak.writeField(myChannelNumber, 1, myBPM, myWriteAPIKey);

    //uncomment if you want to get temperature in Fahrenheit

    //int x = ThingSpeak.writeField(myChannelNumber, 1, temperatureF, myWriteAPIKey);

    if(x == 200){

      Serial.println("Channel update successful.");

    }

    else{

      Serial.println("Problem updating channel. HTTP error code " + String(x));

    }

    lastTime = millis();

  }

}

```

And here's the compile time error that I'm facing rn:

```

C:\Users\DELL\AppData\Local\Temp\arduino-sketch-CBF3B8ED5F9B2285CB06B657267B11A1\libraries\PulseSensor_Playground\PulseSensorPlayground.cpp.o:(.literal._ZN21PulseSensorPlayground5beginEv+0x0): undefined reference to `PulseSensorPlayground::UsingInterrupts'
C:\Users\DELL\AppData\Local\Temp\arduino-sketch-CBF3B8ED5F9B2285CB06B657267B11A1\libraries\PulseSensor_Playground\PulseSensorPlayground.cpp.o:(.literal._ZN21PulseSensorPlayground5beginEv+0x8): undefined reference to `PulseSensorPlaygroundSetupInterrupt()'
C:\Users\DELL\AppData\Local\Temp\arduino-sketch-CBF3B8ED5F9B2285CB06B657267B11A1\libraries\PulseSensor_Playground\PulseSensorPlayground.cpp.o: In function `PulseSensorPlayground::begin()':
c:\Users\DELL\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.cpp:285: undefined reference to `PulseSensorPlaygroundSetupInterrupt()'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

```

abdulmoizs502, 2023-01-23:
> 

> @abdulmoizs502 First, please don't post sensitive information like your ssid, password, api key, etc on this site. We are not responsible for compromised credentials.
> 
> The ESP32 is in our target sights for incorporating into the PulseSensor Playground Library. We should have an example sketch this month. I will post on this issue when we have something for you to try. In the mean time, please tell us what ESP32 board you are using? We are designing the example around the ESP32-S2 variant, which is the one that is the most recent and also has the most support.
> 

I'm using ESP-wroom-32. I've attached the picture. I've connected positive pin of the sensor to 3v3 negative pin to GND and signal pin to Ao(GPIO-36) of ESP32.

[image]

biomurph, 2023-01-23:
@abdulmoizs502 
Thanks for sending the picture.
The ESP module you are using is an older version. The datasheet says is is NRND.
https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf
We are developing on the latest ESP32-S2 model, and we're not sure if it will be backwards compatible with the older ESP32 variants. 

Also, you are using a knock-off clone of our PulseSensor hardware. 
I won't be able to troubleshoot any issues you may have with the cloned hardware that you're using, as I don't know what is even on there. 
Please support the original makers of the PulseSensor by purchasing legit parts. You can see the difference in the hardware on our website, PulseSensor.com. We devote time and effort into making sure that our PulseSensor works on platforms, such as the ESP32.

biomurph, 2023-01-26:
@abdulmoizs502 

Just want to let you know that I added an example for the ESP32 in the examples folder.
Please go to the Releases tab
https://github.com/WorldFamousElectronics/PulseSensorPlayground/releases
Download the latest 1.6.3 release as a zip, and unzip it into your Arduino > libraries folder.

Please let me know if this will work for you? In the example, I am targeting the ESP32-S2 but it should work for the earlier models of ESP32 as well.

Also, please purchase legit parts for the best experience!

chima-okwara, 2023-03-06:
Hello, @biomurph .
I am also getting this same error, but I'm using an STM32F401CDU6 based board, unlike OP. The code doesn't even compile.

Here's a screenshot of the error message:
[image]

biomurph, 2023-03-06:
@chima-okwara please tell me what board you are using that has the STM32F401 on it?

Also, please try using the BPM_Alternative sketch and see if that compiles?

chima-okwara, 2023-03-06:
@biomurph Thanks for the quick response.

I am using a blackpill by WeAct studios. 
[image] 

I'll run the sketch and get back in a minute.

chima-okwara, 2023-03-06:
@biomurph 

I still get an error. This one. 
[image]

biomurph, 2023-03-06:
@chima-okwara 
I see what the problem is. the ST Micro board file use a macro called OUTPUT_TYPE, and we use the same name for a variable in our examples.
First, delete the PulseSensor Playground library that you have, and download the latest release v1.7.0 here
https://github.com/WorldFamousElectronics/PulseSensorPlayground/releases

The PulseSensor_BPM_Alternative.ino example will compile, but you have to change the variable name.
Try changing these two lines

const int OUTPUT_TARGET = SERIAL_PLOTTER;

pulseSensor.setOutputType(OUTPUT_TARGET);

chima-okwara, 2023-03-08:
Hi @biomurph , Sorry for the late response. I returned late from work. When you say to change those lines, what am I to change them to? I already tried running the example and got the following error.

[image]

abdulmoizs502, 2023-03-08:
Change the const int OUTPUT_TYPE = SERIAL_PLOTTER;  line to  = const int OUTPUT_TARGET = SERIAL_PLOTTER;

Similarly ,
change
pulseSensor.setOutputType(OUTPUT_TYPE); line to pulseSensor.setOutputType(OUTPUT_TARGET);

biomurph, 2023-03-08:
If you are still getting that error? It might be due to the board you are selecting?

I compiled against the Blackpill F411CE. What board did you select?

chima-okwara, 2023-03-25:
Hello, @biomurph . Sorry for the late response, I've been occupied with loads of projects. 

I'm using the Blackpill F401CD board.

chima-okwara, 2023-03-26:
Hello @biomurph  and @abdulmoizs502 . I'm sorry for the late response. 
I did compile the code, making the changes as recommended. It compiled successfully. But then, when I ran the code, I got no response from the system. No output from the serial monitor, with my finger firmly pressed against the sensor. 

[image]

biomurph, 2023-03-26:
@chima-okwara 
It is strange that you are not getting a serial output.
Could you please share your full code? Also, your OS and the version of Arduino you are using?

chima-okwara, 2023-03-28:
Hullo @biomurph , sorry for the late response. 
I think I finally figured out why the serial port wouldn't give output. I finally got it to work. Thanks so much for the help. What I did was modify the code to use another serial port, i.e. Serial2 instead of the default Serial port. Here is a screenshot of the output:
[image]

Here is the full code I ran that compiled.
`
/*
   Sketch to handle each sample read from a PulseSensor.
   Typically used when you don't want to use interrupts
   to read PulseSensor voltages.

   Here is a link to the tutorial that discusses this code
   https://pulsesensor.com/pages/getting-advanced

   Copyright World Famous Electronics LLC - see LICENSE
   Contributors:
     Joel Murphy, https://pulsesensor.com
     Yury Gitman, https://pulsesensor.com
     Bradford Needham, @bneedhamia, https://bluepapertech.com

   Licensed under the MIT License, a copy of which
   should have been included with this software.

   This software is not intended for medical use.
*/

/*
   Every Sketch that uses the PulseSensor Playground must
   define USE_ARDUINO_INTERRUPTS before including PulseSensorPlayground.h.
   Here, #define USE_ARDUINO_INTERRUPTS false tells the library to
   not use interrupts to read data from the PulseSensor.

   If you want to use interrupts, simply change the line below
   to read:
     #define USE_ARDUINO_INTERRUPTS true

   Set US_PS_INTERRUPTS to false if either
   1) Your Arduino platform's interrupts aren't yet supported
   by PulseSensor Playground, or
   2) You don't wish to use interrupts because of the side effects.

   NOTE: if US_PS_INTERRUPTS is false, your Sketch must
   call pulse.sawNewSample() at least once every 2 milliseconds
   to accurately read the PulseSensor signal.
*/
#define USE_ARDUINO_INTERRUPTS false
#include <PulseSensorPlayground.h>

/*
   The format of our output.

   Set this to PROCESSING_VISUALIZER if you're going to run
    the Processing Visualizer Sketch.
    See https://github.com/WorldFamousElectronics/PulseSensor_Amped_Processing_Visualizer

   Set this to SERIAL_PLOTTER if you're going to run
    the Arduino IDE's Serial Plotter.
*/
const int OUTPUT_TARGET = SERIAL_PLOTTER;
/*
   Pinout:
     PULSE_INPUT = Analog Input. Connected to the pulse sensor
      purple (signal) wire.
     PULSE_BLINK = digital Output. Connected to an LED (and 1K series resistor)
      that will flash on each detected pulse.
     PULSE_FADE = digital Output. PWM pin onnected to an LED (and 1K series resistor)
      that will smoothly fade with each pulse.
      NOTE: PULSE_FADE must be a pin that supports PWM. Do not use
      pin 9 or 10, because those pins' PWM interferes with the sample timer.
     THRESHOLD should be set higher than the PulseSensor signal idles
[... 95 lines cut ...]
      pulseSensor.outputSample();

      /*
         At about the beginning of every heartbeat,
         report the heart rate and inter-beat-interval.
      */
      if (pulseSensor.sawStartOfBeat()) {
        pulseSensor.outputBeat();
      }
    }

    /*******
      Here is a good place to add code that could take up
      to a millisecond or so to run.
    *******/
  }

  /******
     Don't add code here, because it could slow the sampling
     from the PulseSensor.
  ******/
}
`

Once again, thanks for the help, man.

chungpin092, 2023-06-24:
I have same qustion on ESP32CAM. I use democode can upload but can't see it work

biomurph, 2023-06-26:
@chungpin092 
Please try the code that @chima-okwara used above.
Also, please provide more information in your question to aid in troubleshooting.

biomurph, 2023-10-26:
PulseSenor Playground Library supports ESP32 based boards

closed 2023-10-26 by biomurph