WorldFamousElectronics/PulseSensorPlayground #169: PulseSensr_BPM_Alternative Compilation Errors

edpark, 2023-02-22:
Hi, is anyone able to compile the PulseSensr_BPM_Alternative sketch on these boards?

- [Adafruit ItsyBitsy M4 Express](https://www.adafruit.com/product/3800)
- [Adafruit Feather nRF52840 Express](https://www.adafruit.com/product/4062)
- [Adafruit Feather STM32F405 Express](https://www.adafruit.com/product/4382)

I'm getting different compilation errors like these:

```
In file included from c:\Users\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.h:447,
                 from C:\Users\AppData\Local\Temp\.arduinoIDE-unsaved2023122-10456-z9gy6n.9tjie\PulseSensor_BPM_Alternative\PulseSensor_BPM_Alternative.ino:41:
c:\Users\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundDisableInterrupt()':
c:\Users\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:306:1: error: no return statement in function returning non-void [-Werror=return-type]
  306 | } // PulseSensorPlaygroundDisableInterrupt
      | ^
In file included from c:\Users\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.h:447,
                 from C:\Users\AppData\Local\Temp\.arduinoIDE-unsaved2023122-10456-z9gy6n.9tjie\PulseSensor_BPM_Alternative\PulseSensor_BPM_Alternative.ino:41:
c:\Users\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundEnableInterrupt()':
c:\Users\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:362:1: error: no return statement in function returning non-void [-Werror=return-type]
  362 | }
      | ^
cc1plus.exe: some warnings being treated as errors

exit status 1
```
or
```
In file included from C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\system/STM32F4xx/stm32f4xx_hal_conf_default.h:298,
                 from C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\system/STM32F4xx/stm32f4xx_hal_conf.h:13,
                 from C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:29,
                 from C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:287,
                 from C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\cores\arduino/stm32/stm32_def.h:34,
                 from C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\cores\arduino/stm32/clock.h:19,
                 from C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\cores\arduino/wiring_time.h:23,
                 from C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\cores\arduino/wiring.h:38,
                 from C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\cores\arduino/Arduino.h:36,
                 from C:\Users\AppData\Local\Temp\arduino-sketch-FC8383D349840B19E631B8BA0C1A7831\sketch\PulseSensor_BPM_Alternative.ino.cpp:1:
C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_gpio.h:261:50: error: expected unqualified-id before numeric constant
  261 | #define OUTPUT_TYPE                             (0x1UL << OUTPUT_TYPE_Pos)
      |                                                  ^~~~~
C:\Users\AppData\Local\Temp\.arduinoIDE-unsaved2023122-10456-z9gy6n.9tjie\PulseSensor_BPM_Alternative\PulseSensor_BPM_Alternative.ino:53:11: note: in expansion of macro 'OUTPUT_TYPE'
   53 | const int OUTPUT_TYPE = SERIAL_PLOTTER;
      |           ^~~~~~~~~~~
C:\Users\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.4.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_gpio.h:261:50: error: expected ')' before numeric constant
  261 | #define OUTPUT_TYPE                             (0x1UL << OUTPUT_TYPE_Pos)
      |                                                 ~^~~~~
C:\Users\AppData\Local\Temp\.arduinoIDE-unsaved2023122-10456-z9gy6n.9tjie\PulseSensor_BPM_Alternative\PulseSensor_BPM_Alternative.ino:53:11: note: in expansion of macro 'OUTPUT_TYPE'
   53 | const int OUTPUT_TYPE = SERIAL_PLOTTER;
      |           ^~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

And I take it that hardware interrupts aren't supported on these boards? (I read a suggestion on using this library https://github.com/khoih-prog/NRF52_TimerInterrupt - has anyone managed to get that to work?)

biomurph, 2023-02-22:
@edpark 
please download the latest library from the main github page. we have example for the nRF52 that works with hardware timers.

we are making big improvements this winter and spring. those other two platforms are on the list.

edpark, 2023-02-22:
Thank you @biomurph, I'll try it out and report back later today.

edpark, 2023-02-23:
Wanted to let you know that the `PulseSensor_nRF52 `example compiles and runs on my nRF52840 Express board, thank you! 
For anyone else coming across the same problem, this is what I had to do:

1. Download the latest PulseSensorPlayground zip (https://github.com/WorldFamousElectronics/PulseSensorPlayground/archive/refs/heads/master.zip)
2. Install the library in Arduino IDE by going to the menu `Sketch` > `Include Library` > `Add .ZIP Library` and select the zip file (more detailed instructions here: https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries)
3. Download the NRF52_TimerInterrupt zip file (https://github.com/khoih-prog/NRF52_TimerInterrupt/archive/refs/heads/main.zip)
4. Install the library in the Arduino IDE as in step 2.
5. To open the PulseSensor_nRF52 example, unzip the master.zip file from step 1; the sketch is located under `examples/PulseSensor_nRF52`

closed 2023-02-23 by edpark

biomurph, 2023-02-23:
@edpark Thanks for confirming that it will work!

I have an example that targets the Adafruit nRF52840 Express which advertises HRM service and can be read by the Nordic Toolbox, as well as a handfull of HRM related apps available.

I would add to your number 5 above that you can access the PulseSensor nRF52 example from the File > Examples > PulseSensor Playground as well.

biomurph, 2023-03-05:
@edpark 
Just want to let you know that the latest library in this repository has an example for the nRF52840 Feather Express.
We have not made an official release yet, so you will need to download the zip.

edpark, 2023-03-06:
@biomurph - thanks, will check out!

biomurph, 2023-03-06:
The latest release, v1.7.0 is up to date with all the changes.
It might not show up in the Arduino Library Manager, so you may need to download from our Releases page