WorldFamousElectronics/PulseSensorPlayground #174: Can't compile even when including "USE_ARDUINO_INTERRUPTS true"

jbiribi, 2023-06-01:
I included these two lines:
#define USE_ARDUINO_INTERRUPTS true
#include <PulseSensorPlayground.h>

I'm getting the following code error:

In file included from C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.h:474:0,
                 from C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:16:
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp: In member function 'void PulseSensor::setThreshold(int)':
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:73:45: error: 'cli' was not declared in this scope
 #define DISABLE_PULSE_SENSOR_INTERRUPTS cli()
                                             ^
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:79:3: note: in expansion of macro 'DISABLE_PULSE_SENSOR_INTERRUPTS'
   DISABLE_PULSE_SENSOR_INTERRUPTS;
   ^
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:74:44: error: 'sei' was not declared in this scope
 #define ENABLE_PULSE_SENSOR_INTERRUPTS sei()
                                            ^
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:82:3: note: in expansion of macro 'ENABLE_PULSE_SENSOR_INTERRUPTS'
   ENABLE_PULSE_SENSOR_INTERRUPTS;
   ^
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp: In member function 'boolean PulseSensor::sawStartOfBeat()':
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:73:45: error: 'cli' was not declared in this scope
 #define DISABLE_PULSE_SENSOR_INTERRUPTS cli()
                                             ^
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:107:3: note: in expansion of macro 'DISABLE_PULSE_SENSOR_INTERRUPTS'
   DISABLE_PULSE_SENSOR_INTERRUPTS;
   ^
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:74:44: error: 'sei' was not declared in this scope
 #define ENABLE_PULSE_SENSOR_INTERRUPTS sei()
                                            ^
C:\Users\jonas\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:110:3: note: in expansion of macro 'ENABLE_PULSE_SENSOR_INTERRUPTS'
   ENABLE_PULSE_SENSOR_INTERRUPTS;
   ^

exit status 1

Compilation error: exit status 1

biomurph, 2023-06-01:
@jbiribi 

What hardware platform are you targeting?
Can you describe your setup? A picture of your hardware will help too.

jbiribi, 2023-06-01:
This is my computer: https://imgur.com/rGZN7X5

biomurph, 2023-06-02:
@jbiribi 
What Arduino board are you using?

jbiribi, 2023-06-02:
I'm using arduino due, but this is happening before using it, because it isn't compiling

biomurph, 2023-06-02:
@jbiribi We don't support hardware interrupts for DUE yet.
Try Using the example: PulseSensor_BPM_Alternative.ino

We are actively improving the library. If you make any improvements of your own in the direction of interrupt support for DUE, please send a pull request~

jbiribi, 2023-06-25:
@biomurph With PulseSensor_BPM_Alternative.ino I'm getting many strange symbols like �J�JY�Y�Y���������J�J�J�JY�Y�Y�Y�Y�Y������[�[Ґ��J�J��Y�Y������������������H��Y��HY�Y.
Do you know what could be happening?

biomurph, 2023-06-26:
@jbiribi 
Please check to make sure that your serial port baud rates are set correctly.
The Arduino board and the computer serial monitor need to be set to the same baud, or else you get the kind of garbage that you are seeing.

jbiribi, 2023-06-28:
@biomurph It worked, thanks. 

Is there an example of "Getting_BPM_to_Monitor" without using hardware interrupts?

biomurph, 2023-06-28:
@jbiribi 
There is not an example in our library of "Getting_BPM_to_Monitor.ino" without interrupts.
You could easily modify the BPM_Alternative example sketch to make it do the things that you need.

jbiribi, 2023-06-28:
@biomurph
I tryed to modify it last week, but didn't work. Is it possible to use "pulseSensor.getBeatsPerMinute()" without hardware interrupts?

biomurph, 2023-06-28:
Yes, it is possible.
When you make modifications, go step by step and test the code as you go.

jbiribi, 2023-07-02:
> @biomurph
Unfortunately I can't go testing step by step because BPM to Monitor don't work for me since the beginning because it uses interrupts. I tryed to modify Alternative to be like BPM to Monitor, but nothing is showed on "pulseSensor.getBeatsPerMinute()".
I don't know if you saw someone doing it without interrupts over internet. 
Anyway, thank you, I will try to find how to make it work.

biomurph, 2023-07-07:
I am bumping DUE compatibility up on the list of target boards.

biomurph, 2023-10-30:
@jbiribi I just published a release that supports Arduino DUE. 
As of v1.9.2
Please check your version and test it if you are still on this project.

closed 2023-10-30 by biomurph