WorldFamousElectronics/PulseSensorPlayground #181: No BPM readings

jbiribi, 2023-08-05:
I'm using "Getting_BPM_to_Monitor" code on my original Arduino Uno, but I'm getting no readings. Only get reading if I touch the back of the sensor, where the circuit is (but of course, not correct readings). I bought a second pulse sensor but the problem persists. I tried on another Arduino Uno (not original, but works for other aplications) and no readings too.

biomurph, 2023-08-08:
@jbiribi 
Please send a photo of your hardware setup?  It may be something simple.

jbiribi, 2023-08-09:
@biomurph 
https://imgur.com/5ECDSna

biomurph, 2023-08-10:
@jbiribi 
Thank you for sending the picture.
I can tell by seeing the color of the cable, and the length of the cable, and the PCB, that the sensor that you got is not one of our original PulseSensors. You have purchased a clone copy which doesn't work. There is no way for me to troubleshoot your problem, because I have no idea how the piece is made. Please purchase legit parts from the original makers to ensure functionality and to get good troubleshooting feedback.

closed 2023-10-26 by biomurph