WorldFamousElectronics/PulseSensorPlayground #185: Problem about sensor when connecting with Arduino NANO 33 loT

LinkWang-web, 2023-11-30:
Arduino NANO 33 loT sometimes causes more noise on the plotter monitor in IDE. The input value keeps weaving from 70 to 150. Adjusting the bandage of the sensor would help a bit.

biomurph, 2023-11-30:
@LinkWang-web 
We are aware of the issue with Arduino NANO 33 platform.
Please use the PulseSensor_BPM_Alternative.ino sketch.

Could you please post a screenshot of the noise you are referring to?
Also, your OS and a picture of your PulseSensor setup?

biomurph, 2024-01-05:
@LinkWang-web 
please switch the Playground repository branch to the 'Version-2.0-beta' branch.
Then, delete the library you have installed, and download (or fork) the Version-2.0-beta branch into your Arduino/libraries folder.
Try that and let us know how it works?

biomurph, 2024-03-05:
This is closed because V2 solves the problem

closed 2024-03-05 by biomurph