WorldFamousElectronics/PulseSensorPlayground #188: Pulse sensor BPM_to_Monitor script does not work on Arduino Nano Every

GGExploder, 2023-12-18:
To be clear,
Everything works on the Arduino Uno R3, GettingStartedProject AND BPM_to_Monitor.

GettingStartedProject also works on Arduino Nano Every, it seems specifically the PulseSensorPlayground library is not working and therefore the code isn't getting past 
pulseSensor.analogInput(PulseWire);

This occurs with both the ATMEGA 328 emulation and the ATMEGA 4809.

Any and all help would be greatly appreciated.

GGExploder, 2023-12-18:
For further info, the no interrupt script works fine, so I suspect that the playground doesn't support this chipset's interrupts. Oh well :(

GGExploder, 2023-12-18:
Confirmed, by removing the playground's library the code does work, but less well than I'd like. It'll have to do but an update to the library would always be greatly appreciated! The code is below:
`#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <Fonts/FreeSerif9pt7b.h>

// Screen Definitions
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
Adafruit_SSD1306 Display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);

// Variables
const int PulseWire = A2;  // Analog pin for PulseSensor
const int LED = LED_BUILTIN;
int Threshold = 550;

unsigned long lastBeatTime = 0;
unsigned long thisBeatTime = 0;

void setup() {
  Serial.begin(115200);
  delay(250);
  Display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
}

void loop() {
  int sensorValue = analogRead(PulseWire);

  if (sensorValue > Threshold) {
    thisBeatTime = millis();
    int beatInterval = thisBeatTime - lastBeatTime;

    if (beatInterval > 0) {
      int myBPM = 60000 / beatInterval;

      // Check if BPM is within a reasonable range
      if (myBPM > 40 && myBPM < 200) {
        Display.clearDisplay();
        Display.setFont(&FreeSerif9pt7b);
        Display.setTextColor(SSD1306_WHITE);
        Display.setTextSize(1);
        Display.setCursor(30, 48);
        Display.print("BPM: ");
        Display.println(myBPM);
        Display.display();
      }
    }

    lastBeatTime = thisBeatTime;
  }

  delay(10);  // Adjust the delay as needed for your application
}`

biomurph, 2023-12-18:
Hey @GGExploder 
Yes, you have found a bug in our library that we are actively working to fix.
If you don't mind, please switch the Playground repository branch to the **'Version-2.0-beta'**  branch.
Then, delete the library you have installed, and download (or fork) the **Version-2.0-beta** branch into your `Arduino/libraries` folder.

What we're trying to do is solve for your exact problem. If you can do that, and have some success, we would like to hear!

TLDR:

The current paradigm has placed the user (you) in the position of knowing if the board works with interrupts. That is a pain point we are trying to fix with version 2. In version 2, the library handles all of the hardware timer needs, and just works. Later, as you develop your project, the issue of interrutps v not-interrupts may or may not be an problem. We are always working toward inclusion of hardware targets into our hardware timer family. 

For some reason, the Arduino Nano family, especially what I have come to find as the EMBED family, do not play well with the same library that works seemlessly with Adafruit and Seeed Studio procuced boards (and, I assume, board files and core libraries, etc.). We are working on it. If you want to work on it, please do! Send us a pull request!

For now, the new version 2.0 is designed to 'just work' and you should try it.

biomurph, 2024-01-05:
@GGExploder 
Did you try the v2 branch?

biomurph, 2024-03-05:
This is closed because V2 solves the problem

closed 2024-03-05 by biomurph