WorldFamousElectronics/PulseSensorPlayground #189: example for esp8266

Perry2001, 2023-12-28:
no example for esp8266

biomurph, 2023-12-28:
@Perry2001 
We don't have a specific example for ESP8266 _yet_.
The PulseSensor Playground is a work in progress, and we are always making progress!
We have a version 2.0 release of the library coming up soon. For a preview, please switch from the main branch to the "Version_2.0-Beta" branch, download the .zip and extract it to your Arduino/libraries folder.
This new version takes the headaches out of hardware timers, and our basic example PulseSensor_BPM.ino will just work for every platform.

In regard to a ESP8266 specific example, we are working on it. Here is how it will go:
There is a hardware timer interrupt library for the ESP8266 called [ESP8266TimerInterrupt](https://www.arduino.cc/reference/en/libraries/esp8266timerinterrupt/), which we will use first to build a prototype example. This library is made by the same folks who made the hardware timer interrupt library we use for the ESP32 example that we have. An example built with the same structure as the one for the ESP32, along with necessary changes to the library components, will be tested and then included in a new release. 

Since we are preparing to make 2.0 a major update, we will try to fit in as much board support as possible. I am sure the ESP8266 will be in there.

biomurph, 2024-01-05:
@Perry2001 
Did you try to use our version 2.0 branch of the PulseSensor Playground Library?

I am pushing changes to the branch that provide support for the ESP8266. I am developing on an Adafruit Feather HUZZAH.
The library comiles and runs on ESP8266 using our PulseSensor_BPM.ino example.

The ESP8266 is a very special device. It only has 1 analog pin, and it can only read between 0V and 1V, so you are going to have to divide down the signal that is coming from the PulseSensor. Please be aware of that.
The ESP8266 runs on 3.3V, so a divider circuit using 4.7K and a 2.2K would make the highest PulseSensor value at just a hair above 1V. 

Since the ESP8266 is so special in this way, we will likely have to put together a little tutorial on how to lash it up.

Please let us know how this is working for you? Please use the latest v2 branch of this repository for testing.

biomurph, 2024-03-05:
This is closed because V2 solves the problem