WorldFamousElectronics/PulseSensorPlayground #190: sampleTimer.stop() and sampleTimer.start issues

Marcus-Peterson, 2024-01-10:
I am using Arduino Rev4 and I am trying to use the code: Getting_BPM_to_Monitor.ino as provided with the library example. But I keep getting this error:

(FYI I am using generic placeholders for paths as to not expose sensitive information)

`In file included from [user path]\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.h:486:0,
                 from [temporary path]\Getting_BPM_to_Monitor\Getting_BPM_to_Monitor.ino:13:
[user path]\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundDisableInterrupt()':
[user path]\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:328:5: error: 'sampleTimer' was not declared in this scope
     sampleTimer.stop();
     ^~~~~~~~~~~
In file included from [user path]\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.h:486:0,
                 from [temporary path]\Getting_BPM_to_Monitor\Getting_BPM_to_Monitor.ino:13:
[user path]\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundEnableInterrupt()':
[user path]\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:402:5: error: 'sampleTimer' was not declared in this scope
     sampleTimer.start();
     ^~~~~~~~~~~

exit status 1

Compilation error: exit status 1
`

Marcus-Peterson, 2024-01-10:
For some reasons, markdown of code isn't working

biomurph, 2024-01-10:
@Marcus-Peterson 
We are getting ready to push a new version of the PulseSensor Playground.
Please remove the version of the library that you have, and download-install the branch called 'Version 2.0 Beta'.

Please let us know how that goes?

Marcus-Peterson, 2024-01-10:
> @Marcus-Peterson We are getting ready to push a new version of the PulseSensor Playground. Please remove the version of the library that you have, and download-install the branch called 'Version 2.0 Beta'.
> 
> Please let us know how that goes?

Sure, I'll do that

Marcus-Peterson, 2024-01-11:
> @Marcus-Peterson We are getting ready to push a new version of the PulseSensor Playground. Please remove the version of the library that you have, and download-install the branch called 'Version 2.0 Beta'.
> 
> Please let us know how that goes?

Yes, using the version 2.0 beta seemed to work fine. For anyone who might stumble onto this git issue through a google search. Either you can do what I did and comment out the sample functions or just use the version 2.0. 

Either way, it's going to work.

Ohh, and another thing. Use 3 volts and not 5 volts for the pulse sensor. Trust me on this one.

closed 2024-01-11 by Marcus-Peterson

biomurph, 2024-01-12:
@Marcus-Peterson 
Thanks for the feedback, and thanks for using PulseSensor!
We will be releasing the v2 in a day or so.

Please do post an issue if you have found that you are getting a better signal with 3V over 5V? We want to make sure that our customers are aware of easy SNR reduction options.