WorldFamousElectronics/PulseSensorPlayground #192: `multiple definition of `__vector_11'` when including `#include <Servo.h>`

gohai, 2024-02-22:
Including `<Servo.h>` produces this linker error with any of the PulseSensorPlayground example sketches.
This is with Servo version 1.1.8 (as part of Arduino 2.3.2).

gohai, 2024-02-22:
The error goes away if I change

```
      #if defined Servo_h
```
to
```
      #if 1
```

in `TimerHandler.h`. So this detection seems to be broken with the given (current) Arduino version.

biomurph, 2024-02-22:
@gohai We have not noticed this before. It is quite unexpected, as we have recently upgraded the library in coordination with our book and tested all the examples. I have replicated your issue, and I find that the compilation error is limited to Arduino AVR and other AVR boards. This compilation error does not occur when using SAM architecture, or Arduino R4, for example.

I am using the Servo library by Michael Margolis, which is called Servo if you search to install it. It appears to be cloned in the Arduino15 libraries, and they both have this error at compile time. The issue is with AVR and using Timer 1, which is used by the Servo library. Our precompile directives steer the compiler to Timer 2 for the UNO, but somehow the code for Timer 1 is still getting hit. I tried your alteration of the precompile macro, and I also see that it will compile without errors. I also just commented out the section that sets up Timer 1, and again, it compiles without errors. 
This is strange, because the compiler directives are supposed to steer clear of this confict.

Can you confirm that changing 
`#if defined Servo_h `
to 
`#if 1`
and then uploading to Arduino UNO (ATmega328P) will run and drive the servo as expected?

biomurph, 2024-02-22:
I have verified that the TimerHandler.h section that tests to see if the Servo.h library is included is not detecting it.
Our test is 
`#if defined Servo_h`
We make note to include the Servo.h library in the code prior to including PulseSensor Playground for this very purpose. The compiler should include Servo.h, which has the header guard
`#ifndef Servo_h`
`#define Servo_h`
So somehow the PulseSensor Playground test for pre-inclusion of Servo.h is not getting hit. That should not be happening.

gohai, 2024-02-23:
Thank you for looking into this @biomurph.

I _believe_ what's happening is that TimerHandler.h is in a separate translation unit than main sketch. `Servo_h` does get defined (in the main sketch, when that header file gets pulled in). But it won't be defined when the `.cpp` files of PulseSensor Playground get compiled, which happens independently (and the two are only getting joined at link-time, which is when this this error gets thrown). See also [here](https://forum.arduino.cc/t/compiler-not-detecting-external-header-file-with-defines/526633/6).

I should be able to test on Tuesday if the Servo functionality works after the change.

biomurph, 2024-02-23:
@gohai Yeah, I actually saw that thread and I do believe you are correct in that the error is happening at link time.
This is tough, because we have to time our inclusion of TimerHandler.h in such a way as to ensure that other stuff is set up and working.
I have tried a few things, but they all point to the problem being in the linker stage.
This may take a bit to fixer...

biomurph, 2024-02-26:
@gohai 
The trouble with moving TimerHandler.h into a new position is that it needs to be able to call our function that happens on timer interrupt. That said, it would be in the PulseSensorPlayground.h where this specific `#ifdef` arbitration needs to happen. The easiest way I can see to manage it is to have two separate H files that do the TimerHandler stuff based on the inclusion of Servo.h or not. That will be tested this week, and a new version of the library will get posted as a new release.

FWIW, the reason we have been doing this, that is, selecting different timers in the event Servo.h is included, is because there is the potential for users to program in the tone() function, which would cause a collision. The use of tone() is not knowable during compilation, so we went with checking for Servo.h to reduce the chance of collision as much as possible.

gohai, 2024-02-28:
Thank you @biomurph.

FWIW: We tested adding `#define Servo_h` at line 19 of PulseSensorPlayground.cpp, and this made both servo and pulse sensor work (with Arduino Uno).

biomurph, 2024-02-28:
@gohai
This issue only applies to AVR micros, and including the Arduino MEGA architecture.
So far, that is the only family that I have found this issue arising with.

It makes sense that adding `#define Servo_h` in the cpp would work.
However, that is not the best or most elegant solution.
Due to the structure of our library, we 'loose sight' of the definition of Servo_h during compilation and linking.
In order to solve this problem, I went down a couple of rabbit holes, but the last rabbit hole had the rabbit in it!

I use the conditional prior to selecting an appropriate timer for AVR architecture
`#if __has_include (<Servo.h>)`
This is working to compile correctly. Yay!
I have tested on an UNO with a servo motor. 

Please, download the zip of this repository (v2.1.0) and test it with your project code? Let us know how it works.
I will be updating the library to the Arduino ecosystem today, but it may take a bit for the libraries manager to pick it up.

biomurph, 2024-02-29:
@gohai Thanks for posting the issue.
Please, when you have a chance to run the new library with your code, Come back here and close this issue if it works as expected?
Have a great day!

XCF-KHX, 2024-03-04:
@biomurph Hi, I tested it out for @gohai and it seems that while the compilation error has been fixed the sensor fails to get any readings unless `#include <Servo.h>`  (as opposed to previously `#define <Servo.h>`) is added to line 19 of PulseSensorPlayground.cpp.

biomurph, 2024-03-04:
@XCF-KHX Thank you for testing and posting.
I had been away from my lab. Now I am back and I was able to replicate your issue.
I had to dig in a bit deeper to chase down the places were inclusion of the servo library is tested.
I have an Arduino UNO lashed up to a servo motor and PulseSensor, and both the PulseSensor input and the Servo output are working as expected in the PulseSensor_Servo_Motor.ino example.

Please download the latest version `v2.1.1` from this repo, or use the Arduino Libraries Manager (might take a bit for the Library Manager to scrape our repo?)

Please let us know if this is the final fix?

XCF-KHX, 2024-03-05:
@biomurph I've just tested with an UNO as well, with v2.1.1 everything seems to be working now. Thanks for all your efforts, have a good day!

closed 2024-03-05 by biomurph