WorldFamousElectronics/PulseSensorPlayground #194: Heart rate is abnormally high

nammoorigebanni68, 2024-05-16:
i have been using heart beat pulse sensor. No matter what, heart beat is over 200

biomurph, 2024-05-16:
@nammoorigebanni68 
Sorry to hear you are having this problem.
Please give us more information so that we can troubleshoot your problem.
What OS are you using?
What microcontroller board are you using?
What code, or  example are you using?
Please also drop in a photo of your setup, so that we can understand your use case better.

shivanjali16rao, 2024-05-25:
> @nammoorigebanni68 Sorry to hear you are having this problem. Please give us more information so that we can troubleshoot your problem. What OS are you using? What microcontroller board are you [...]

hey! 
os - windows 11
microcontroller- arduino uno r3
code - pulsesensorplayground example - getting bpm on monitor
I don't have the actual photo of connections, I will attach the reference we used

[image]

biomurph, 2024-07-18:
@shivanjali16rao 
Please reply with a photo of your PulseSensor setup so that I can help you troubleshoot.
If I don't hear from you by August, I will have to close this issue.

biomurph, 2024-07-29:
This issue is closed due to non-response.