WorldFamousElectronics/PulseSensorPlayground #195: Pulse Sensor cpp library obsolete with Espressif ESP32 core update from versions 2.x to 3.x

ElectricDream5, 2024-07-14:
Pulse Sensor was working well with ESP32S3 until recently when Espressif updated the Arduino ESP32 2.x versions to 3.0 and beyond. 

Apparently the new version has important API changes that affect the Pulse Sensor libraries. For example, 16 timer API's are no longer functional, four new timer API's were created, and others changed the required arguments. As a result, the current Pulse Sensor libraries no longer work on the basic BPM example with ESP32.

Has anyone been able to adapt the timer functions and arguments in the cpp library? I'm not clear what values go with each argument.

For example:

c:\Users\User\Documents\Arduino\libraries\PulseSensor_Playground\src\PulseSensorPlayground.cpp:575:15: error: too few arguments to function 'void timerAlarm(hw_timer_t*, uint64_t, bool, uint64_t)'
575 | timerAlarm(sampleTimer);

biomurph, 2024-07-15:
@ElectricDream5 
Well, that's a bummer but not surprising.
Will have to dig in and see how to revive our timers for ESP boards.
Have you tried to roll back your hardware files? That may be the stop-gap solution until we get a handle on the new API tools.

ElectricDream5, 2024-07-17:
Yep, brand new Espressif core for Arduino comes with big changes, and hopefully these will translate into improvements as we adapt them. 

In the mean time, I did roll back to Espressif Arduino core 2.17, and that works fine. I can upload sketches and the timers work as before.  Adapting to 3.x requires changes to the ways timers and interrupts are called in the libraries, among other changes I imagine.

biomurph, 2024-07-18:
Thank you for doing that testing.
I am out of the office, and don't have an esp board with me, but I will be cranking on the changes today and reply here when I have something to test.

biomurph, 2024-07-18:
@ElectricDream5 
I just pushed changes to the library. The version in the repo is now v2.2.0
Please remove the PulseSensor Playground library that you currently have (v2.1.1) and thendownload this repository as a zip file. Once downloaded, click on `Sketch > Include Library > Add .ZIP Library` to get it into your Arduino software.

Please try this with your ESP32 setup?
I have compiled it successfully but I don't have ready access to an ESP32 board.

ElectricDream5, 2024-07-19:
HI Joel. Thank you for posting a solution so quickly!

Per your instructions, I reinstalled ESP32 v. 3.02 and the new PulseSensor Playground repo version 2.20 on my laptop.  After that I tried the BPM example and my own PulseSensor sketch on a LilyGo T Display ESP32 S3 and a Seeed Xiao Esp32C3 dev board.

Happy to report both sketches are able to compile, and time between heartbeats seems to work just fine. Timer issue solved.

However, on both dev boards, I get the following error when I try the ESP32 example from the 2.2 repo:

c:\Users\User\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
c:\Users\User\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
   74 |   mbedtls_md5_starts_ret(&_ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_starts
c:\Users\User\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:75:3: error: 'mbedtls_md5_update_ret' was not declared in this scope; did you mean 'mbedtls_md5_update'?
   75 |   mbedtls_md5_update_ret(&_ctx, data, len);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_update
c:\Users\User\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:76:3: error: 'mbedtls_md5_finish_ret' was not declared in this scope; did you mean 'mbedtls_md5_finish'?
   76 |   mbedtls_md5_finish_ret(&_ctx, _buf);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_finish
c:\Users\USER\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp: In member function 'void AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*)':
c:\Users\User\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp:189:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  189 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf

exit status 1

Compilation error: exit status 1

biomurph, 2024-07-19:
@ElectricDream5 
Thank you for testing the library functionality!

I think the problem that you are now encountering has to do with the library that you are using.
Turns out, three are a few ESPAsyncWebServer embodiments out there.
Please install the libraries listed below. It 'should' work with your ESP32-S3 hardware?

#include <AsyncTCP.h> // https://github.com/dvarrel/ESPAsyncTCP
#include <ESPAsyncWebServer.h> // https://github.com/dvarrel/ESPAsyncWebSrv

ElectricDream5, 2024-07-20:
While we were going back and forth Espressif released version 3.0.3 of the Arduino core. Timers still look good on both boards. 

I deleted the ESPAsincWebServer library and noticed I already have the two you suggest installed in their latest versions. Still getting an error with the ESP32 PulseSensor Playground ESP32 example, but I will fiddle with it some more here to see if I can get it running.

Thank you again.

biomurph, 2024-07-22:
@ElectricDream5 OK, I am glad that it was short work. 
I will close this issue, and if you have any further problems with the ESP example, please create a new issue. 
Thanks for using PulseSensor!