WorldFamousElectronics/PulseSensorPlayground #199: Filtering messages: IBI, Sample, Beat

manno-xx, 2024-12-19:
Hi!

I'm trying to hook up the pulsesensor to Unity to get the IBI (or HRV), but fail to properly filter out the info that I want. I only need the "Q values" send from the PulseSensor_BPM.ino (right?) but I cannot find out how to only get those messages. 
I currently only have the line:
`pulseSensor.outputBeat();` output anything to serial. The other output lines are commented out.

I'm looking for a way to filter because, using Ardity (https://ardity.dwilches.com/) for the reading, I often get a 'queue is full' message from that library (see screenshot) and funnily enough, almost only 'Q messages' are dropped because of that, and not the 'B messages'.

[image]

Strange thing is that the messages (Q and B combined it seams) only appear just a little bit short of a second after one and other. Nothing Unity should not be able to handle, I think.

Is there any way that I can have the PulseSensorPlayground only send certain messages?

Kind regards,
Manno

manno-xx, 2024-12-19:
Duh! Setting MaxUnreadMessages in Ardity's SerialController increases the Queue's size and stops the dropping. 

Still would be nice if one can only get the data one wants?

biomurph, 2024-12-19:
@manno-xx 
Happy to hear that you figured your problem.
We really do need to make this page more visible.
Please check here
https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/resources/PulseSensor%20Playground%20Tools.md#getinterbeatintervalms
and take a look at the rest of the tools that we have made for you.

Does this help?

manno-xx, 2024-12-20:
Thanks @biomurph, it does!
I must admit that this time I don't think I checked, but maybe add it to the wiki pages?

closed 2024-12-20 by biomurph