WorldFamousElectronics/PulseSensorPlayground #202: Inconsistent Beat Detection Issue with PulseSensor for PTT Measurement

szhu07, 2025-02-21:
I'm writing to report an issue I'm experiencing while using two PulseSensors to measure Pulse Transit Time (PTT). I have the sensors set up with one on the wrist and one on the fingertip, using an Arduino Uno for data collection.
The main issue I'm encountering is inconsistent beat detection between the two sensors. What typically happens is that one sensor gets "stuck" in a beat-detected state while the other sensor fails to detect beats. For example, I'll see the wrist sensor continuously showing as detected while the fingertip sensor shows no detection, or vice versa. Initially, this was partially due to having the sensors connected to different voltage levels (3.3V and 5V), but the issue persists even after correcting this by properly splitting the 5V supply to both sensors using a breadboard.
The signal levels from both sensors appear strong and within normal ranges (wrist: 502-526, fingertip: 431-609), but the beat detection remains inconsistent. I've tried adjusting the detection thresholds (currently at 475 for wrist and 400 for fingertip) and implementing timeout mechanisms to reset stuck detections, but the problem continues.
This inconsistent detection prevents accurate PTT calculation since it requires properly timed beats from both sensors. Could you advise on potential solutions or if there might be other factors I should consider regarding sensor placement, contact pressure, or ambient light effects?
Thank you for your assistance.

szhu07, 2025-02-21:
> [@szhu07](https://github.com/szhu07) Thanks for using PulseSensor in your PTT application. I need more information from you in order to help out.
> 
> * Please post a photo or two of your hardware and setup?
> * Please post a screenshot of the pulse waveforms you are seeing? Arduino plotter or our Processing [sketch](https://github.com/WorldFamousElectronics/PulseSensor_Pulse_Transit_Time) for two PulseSensors.
> 
> The wrist is not a super ideal place to try to get PPG from. It can be done, but the signals on the wrist are not as strong as signals from parts of the body with capillary tissues. Have you tried using fingertip/earlobe for PTT?

Hi! When I try to plot the waveforms with Arduino plotter and the Processing sketch, nothing is showing up. I will send pictures of my hardware connection and you can find the code I'm using in my public repo!

https://github.com/szhu07/pttProject

szhu07, 2025-02-21:
[image]

[image]
[image]
[image]
[image]

These are images of the setup!

szhu07, 2025-02-22:
@biomurph 

This is what it shows when I run the code for PTT visualizing in Processing.

[image]

biomurph, 2025-02-24:
@szhu07 
Please make sure that you are modifying our example sketch so that the data is formatted correctly for the PulseSensor Visualizer.

Look for a line of code that says
`const int OUTPUT_TYPE = `
and make sure it says
`const int OUTPUT_TYPE = PROCESSING_VISUALIZER;`

and make note of the comments in that area of the code.
That will format the data so that it is readable by the visualizer.

I do note that you are using legit PulseSensor hardware. Thank you for supporting the original makers of PulseSensor.

szhu07, 2025-02-25:
@biomurph 

Hi! I just checked, it does say that. However when I do serial plotter it still doesn't show anything.

[image]

biomurph, 2025-02-25:
@szhu07 
That is strange. 
OK instead of using the Processing Visualizer, change the OUTPUT_TYPE back to SERIAL_PLOTTER and run the Arduino Serial Plotter to visualize the PulseSensor data?

szhu07, 2025-02-25:
@biomurph 

Also! When I open the file for Processing that I downloaded from the repo, I couldn't find a line that says 
const int OUTPUT_TYPE = PROCESSING_VISUALIZER; or just const int OUTPUT_TYPE in the code. On Processing, I open the file that I download from the repo, correct?

biomurph, 2025-02-25:
@szhu07 
We have two ways to visualize the heartbeat data

**Arduino Serial Plotter**
In your Arduino Sketch, make sure that the OUTPUT_TYPE = SERIAL_PLOTTER
Then, program the board and open the Arduino Serial Plotter, and you should see PulseSensor data

**PulseSensor Processing Visualizer**
In your Arduino Sketch, make sure that the OUTPUT_TYPE = PROCESSING_VISUALIZER
Then, open the Processing sketch of your choice from our repo. You are  exploring PTT, so when you open https://github.com/szhu07/pttProject you will be given a prompt to connect to your arduino board. Use the same board that you used to program it. If you have Arduino IDE open, make sure you close all serial ports, or else the Processing Sketch won’t be able to find the board.

szhu07, 2025-02-26:
@biomurph 

This is what I am getting in Arduino serial plotter

[image]

However, I'm not seeing a line with OUTPUT_TYPE in Processing that I downloaded for the PTT repo from you guys. Below is the code that I downloaded from your repo.

`/*
DISPLAYS MULTIPLE PULSE SENSOR DATA STREAMS
THIS PROGRAM WORKS WITH PulseSensorAmped_n_Sensors ARDUINO CODE

PRESS 'S' OR 's' KEY TO SAVE A PICTURE OF THE SCREEN IN SKETCH FOLDER (.jpg)
PRESS 'R' OR 'r' KEY TO RESET THE DATA TRACES
MADE BY JOEL MURPHY WINTER 2016, MODIFIED WINTER 2017
UPDATED BY JOEL MURPHY SUMMER 2016 WITH SERIAL PORT LOCATOR TOOL
UPDATED BY JOEL MURPHY WINTER 2017 WITH IMPROVED SERIAL PORT SELECTOR TOOL

THIS CODE PROVIDED AS IS, WITH NO CLAIMS OF FUNCTIONALITY OR EVEN IF IT WILL WORK
      WYSIWYG
      THIS CODE SUBJECT TO CHANGE WITHOUT NOTICE
*/

import processing.serial.*;
PFont font;

Serial port;
int numSensors = 2;

int[] Sensor;      // HOLDS PULSE SENSOR DATA FROM ARDUINO
int[] IBI;         // HOLDS TIME BETWEN HEARTBEATS FROM ARDUINO
int[] BPM;         // HOLDS HEART RATE VALUE FROM ARDUINO
int[][] RawPPG;      // HOLDS HEARTBEAT WAVEFORM DATA BEFORE SCALING
int[][] ScaledPPG;   // USED TO POSITION SCALED HEARTBEAT WAVEFORM
int[][] ScaledBPM;      // USED TO POSITION BPM DATA WAVEFORM
float offset;    // USED WHEN SCALING PULSE WAVEFORM TO PULSE WINDOW
color eggshell = color(255, 253, 248);
int heart[];   // USED TO TIME THE HEART 'PULSE'
int PTT;

//  THESE VARIABLES DETERMINE THE SIZE OF THE DATA WINDOWS
int PulseWindowWidth; // = 490;
int PulseWindowHeight; // = 512;
int PulseWindowX;
int PulseWindowY[];
int BPMWindowWidth; // = 180;
int BPMWindowHeight; // = 340;
int BPMWindowX;
int BPMWindowY[];
int spacer = 10;
boolean beat[];    // set when a heart beat is detected, then cleared when the BPM graph is advanced

// SERIAL PORT STUFF TO HELP YOU FIND THE CORRECT SERIAL PORT
String serialPort;
String[] serialPorts = new String[Serial.list().length];
boolean serialPortFound = false;
Radio[] button = new Radio[Serial.list().length*2];
int numPorts = serialPorts.length;
boolean refreshPorts = false;

void setup() {
  size(900, 725);  // Stage size
  frameRate(50);
  font = loadFont("Arial-BoldMT-24.vlw");
  textFont(font);
  textAlign(CENTER);
  rectMode(CORNER);
  ellipseMode(CENTER);
  // Display Window Setup
  PulseWindowWidth = 490;
[... 190 lines cut ...]
void resetDataTraces(){
  for (int i=0; i<numSensors; i++) {
    BPM[i] = 0;
    for(int j=0; j<BPMWindowWidth; j++){
      ScaledBPM[i][j] = BPMWindowY[i] + BPMWindowHeight;
    }
  }
  for (int i=0; i<numSensors; i++) {
    Sensor[i] = 512;
    for (int j=0; j<PulseWindowWidth; j++) {
      RawPPG[i][j] = 1024 - Sensor[i]; // initialize the pulse window data line to V/2
    }
  }
}

void printDataToScreen(){ // PRINT THE DATA AND VARIABLE VALUES
    fill(eggshell);                                       // get ready to print text
    text("Pulse Sensor Pulse Transit Time Visualizer", 300, 30);     // tell them what you are
    for (int i=0; i<numSensors; i++) {
      text("Sensor # " + (i+1), 800, BPMWindowY[i] + 220);
      text(BPM[i] + " BPM", 800, BPMWindowY[i] +185);// 215          // print the Beats Per Minute
      text("IBI " + IBI[i] + "mS", 800, BPMWindowY[i] + 160);// 245   // print the time between heartbeats in mS
    }
    text("PTT " + PTT + "mS", 800, BPMWindowY[1] - 50);
}`

biomurph, 2025-02-26:
@szhu07 
Please note that the OUTPUT_TYPE variable is only in the Arduino sketch. It is NOT in the Processing sketch.
The value of the variable tells the Arduino sketch how to format the data it sends to the computer. The Processing Sketch is on the computer, and will only accept data that is correctly formatted to it.

Thank you for sending the screenshot of the PulseSensor waveforms.
I am guessing that the red line is the PulseSensor on your fingertip, and the blue line is the PulseSensor on your wrist, correct?

The red trace is saturating the signal range at the high end, that is why you are seeing the ‘flat top’ to the waveform.
You can adjust the amount of pressure to bring the signal down so that the peak is visible within the ADC range (0-1023).

For the blue trace, try different parts of your body to see how the waveform changes. You mentioned that you are using your wrist, but try your upper forearm? other places? See where you get better or worse pulse signals?

szhu07, 2025-02-26:
@biomurph 

What's the reason for why the wrist is a bad spot?

Also do you think it's possible to create a cuffless blood pressure monitor with the two pulsesensors using PTT?

biomurph, 2025-02-28:
@szhu07 
PPG measures the change in tissue density. When the shockwave from your heart beat travels through your body, tissues that are infused with blood will become less dense, and that is what is measured as the ‘pulse’.
Because of this, the best places on your body to get a clean PPG signal is where there is capillary tissue (fingertip, earlobe, etc). The wrist area does not have alot of capillary tissue. Muscle, bone, tendon, but not much capillary tissue.

Please try the PulseSensor on different parts of your body to see where you get good signal?

Your idea to create a cuffless blood pressure monitor is a known thing, however it will not give you an absolute BP measurement. It will only give you relative change in arterial stiffness. There is lots of info online (science publications, white papers) that go into this.

szhu07, 2025-03-02:
@biomurph 

Is the pulse playground one sampling it at 2ms? Can I make it 1?

biomurph, 2026-09-15:
Closed because old issue

closed 2026-09-15 by biomurph

yury-g, 2026-09-16:
Thanks for cleaning this up. We can keep the ongoing PTT testing and discussion in #210.