WorldFamousElectronics/PulseSensorPlayground #203: 2025 - Getting BPM to Monitor Not Working on Arduino Nano 33 BLE

AudreyW17, 2025-04-22:
Hello,

I’m currently working with an Arduino Nano 33 BLE. I have been trying to get a heartbeat detected and get the BPM displayed on my serial monitor. Everything is compiling, however no value is being sent out as an output: The raw values are being recieved (and I’ve been able to print out the raw data values), however there seems to be an issue with the getBeatsPerMinute function. 

I looked through [this entire thread](https://github.com/WorldFamousElectronics/PulseSensorPlayground/issues/140) and have tried several example sketches (Getting_BPM_to_Monitor and GettingStartedProject) provided in the library, but none of these have been able to work. 

The thread mentioned that the provided code is designed to use hardware interrupts, which the Nano 33 BLE does not.  As a workaround, it mentioned using the BPM_alternative.ino sketch, however that is no longer available in the PulseSensor Playground repository.   I also looked at the PulseSensor_nrf52_Heartrate_monitor_service program (also as mentioned in the linked thread) however that program is specifically designed for an Adafruit board and the code in the example provided is different from the code referenced in the thread.

Can you please provide some guidance on how to proceed?

Thanks.

biomurph, 2025-04-25:
@AudreyW17 
Our latest PulseSensor Playground library manages the use of a hardware or software timer for you, so we don’t have the BPM_Alternative sketch any more.

When the software compiles in the Arduino IDE using a software timer, our library will print a ‘warning’ to the terminal that says 
`#warning "Using software timer to sample PulseSensor.”`

This is just a notification so you know. Are you seeing that ‘warning’?

Also, please post a photo of your hardware setup, so that I can see what you’re doing?

AudreyW17, 2025-04-25:
@biomurph 

Thank you for the reply.

I have checked and I do have the latest PulseSensor Playground library (2.3), however the example sketches provided to not print the warning (which you cited above) to the terminal. I was wondering which example sketch had the warning you mentioned. I have checked Getting_BPM_to_Monitor, GettingStartedProject, and PulseSensor_BPM, but perhaps I am not using the appropriate sketch.

Additionally, attached below is a photo of my hardware setup. 

[image]
[image]

biomurph, 2025-04-28:
@AudreyW17 
Thanks for the pictures. I can verify that you are indeed using a legit PulseSensor (there are alot of knockoffs out there).

You may not be seeing the warning unless you activate verbose output.
Go to Arduino > Settings > Show verbose output during  and then click ‘compile’.

I do have a Nano33 BLE here in my lab. I will attempt to replicate your problem today/tomorrow latest.

biomurph, 2025-04-28:
@AudreyW17 

I fixed a bug that was causing a problem with the correct timing when using software timer to sample PulseSensor.

Please update your PulseSensor Playground library to v2.4.0 and try it out?

biomurph, 2025-04-29:
@AudreyW17 
Glad to hear that the fix is working for you.
I will close this issue.

The thing you’re seeing with the high bpm values is very likely due to movement noise.
We are just bags of blood and bone, sloshing around all the time. When you move, sometimes you can change the tissue density under where the PulseSensor is ‘looking’ at. Since PulseSensor (and all PPG devices) measure the change in tissue density to get the pulse, movement can increase noise.

Your best way to troubleshoot is to set up the example sketch so that you can watch the PulseSensor signal on the Arduino Serial Plotter. Then you will be able to see how your signal changes when you move. Also, try different body parts. The earlobe, for example is a good isolated capillary tissue area.